Heeks/heekscad #408: OS X Build and issing toolbar icons

Blokkendoos, 2016-03-26:
Successfully built HeeksCAD and CNC on OS X. Though the GUI shows an incomplete toolbars and missing options' tree. 
Only the dropdown (CFlyout*) items are missing in the toolbars. The BitmapButtons are ok.
Debugging (with GDB) wxWidgets is harder than I expected (I am not a developer).
Attached files show the GUI as well as the wxWidgets alert which accompanies Quitting the App.

Has someone more knowledgeable, of wxWidgets on Mac in particular, a hint for me on how to fix this?
Thanks in advance,
Jo

**Details**
OS X 10.8.5
Macports portfiles:
cmake @3.5.0_0 
OpenCascade CE, macport oce@0.17_0
wxWidgets-3.0 @3.0.2_4 
boost @1.59.0_2+no_single+no_static+python27+universal
Sources used:
opencamlib: https://github.com/aewallin/opencamlib
libarea:  https://github.com/Heeks/libarea.git
heekscad: https://github.com/Heeks/heekscad.git
heekscnc: https://github.com/Heeks/heekscnc.git

[image]
[image]
[image]

Blokkendoos, 2016-04-01:
The missing options tree was caused by a workaround for a missing or invalid font. Workaround for this in propgrid.cpp:2693:
```
    if ( m_font.Ok() )
        // This should be otherwise called by SetOwnFont
	    CalculateFontAndBitmapStuff( wxPG_DEFAULT_VSPACING );
    else if ( wxNORMAL_FONT->Ok() )
    {
        m_font = *wxNORMAL_FONT;
	    CalculateFontAndBitmapStuff( wxPG_DEFAULT_VSPACING );
	}
    else
    {
        wxFont useFont = wxScrolledWindow::GetFont();
        wxScrolledWindow::SetOwnFont( useFont );
    }
```
The missing / misplaced toolbar buttons still remain.
[image]

Blokkendoos, 2016-04-08:
The controls not showing seems to be a [wxWidget idiosyncratic behaviour on OS X](https://forums.wxwidgets.org/viewtopic.php?t=40283). Creating the toolbars as a child of a wxPanel worked!
[image]

closed 2016-04-08 by Blokkendoos

ffurano, 2016-12-14:
Hi,

 Brilliant, I was just having the same issues as you. I recently cloned, though, and I don't see your fixes. Didn't you do a pull request ?

 Moreover, the fix for the propgrid.cpp:2693 in the source I have was at line 3693

 Being completely ignorant on wxWidgets, what do you mean with "Creating the toolbars as a child of a wxPanel worked" ? I see that the toolbars are created in HeeksFrame.cpp, what did you do there ?