Heeks/heekscad #410: Installation on Ubuntu 16.04

m-paniez9292, 2016-06-08:
Hi,
I would like to install heekscad on Ubuntu 16.04, but I have some trouble with dependances :
- I can't install liboce-foundation8, but I can install liboce-foundation10 (same with modeling and visualization)
- I can't install libwxbase2.8, but I can install libwxbase3.0-0v5 (same with wxgtk).

So, how can I finish install ?

Thanks

acolomb, 2016-06-08:
I experienced the same troubles. Seems like the Xenial build of current Heeks*** in the PPA failed. That's why the dependencies from the old version are being requested, but are no longer available in Xenial.

acolomb, 2016-06-08:
As a workaround, you could manually download and install the needed .deb's from Launchpad, they were still available in Wily.

I have them leftover from the Wily - Xenial upgrade and it seems to work OK.

neomilium, 2016-06-22:
We could not provide PPA for Xenial (16.04) due to bugs in OCE packages in Ubuntu (and Debian).
https://bugs.launchpad.net/ubuntu/xenial/+source/oce/+bug/1556680

Fixes are now committed, and we want their release upstream in Xenial.

It should took few weeks.

neomilium, 2016-06-30:
Patches have been applied in Xenial.
Enjoy !

closed 2016-06-30 by neomilium