Heeks/heekscad #411: Start failed after build on archlinux with segmentation (core dumpbed)

NeodarZ, 2016-06-30:
I tried to build heekscad with this method command :
```
$ export CASROOT=/opt/opencascade
$ mkdir -p build
$ cd buid
$ cmake -DCMAKE_INSTALL_PREFIX=/usr ..
$ make DESTDIR="$pkgdir/" install
```

But when I would start heekscad, I see this error : 

> 21:02:06: Warning: Mismatch between the program and library build versions detected.
> The library used 3.0 (wchar_t,compiler with C++ ABI 1009,wx containers,compatible with 2.8),
> and your program used 3.0 (wchar_t,compiler with C++ ABI 1010,wx containers,compatible with 2.8).
> 09:02:06 PM: Debug: Failed to connect to session manager: SESSION_MANAGER environment > variable not defined
> 09:02:06 PM: Debug: Adding duplicate image handler for 'PNG file'
> Attempt to automatically load HeeksCNC: /usr/lib/libheekscnc.so
> [1]    14388 segmentation fault (core dumped)  heekscad

neomilium, 2016-07-06:
A debug trace could help :-)

Nevertheless, the warning could point the problem : 
~~~
Mismatch between the program and library build versions detected.
The library used 3.0 (wchar_t,compiler with C++ ABI 1009,wx containers,compatible with 2.8),
and your program used 3.0 (wchar_t,compiler with C++ ABI 1010,wx containers,compatible with 2.8).
~~~

jb36, 2016-09-12:
Try commenting out these lines:

```
diff --git a/src/HeeksCAD.cpp b/src/HeeksCAD.cpp
index 5292107..26215fe 100644
--- a/src/HeeksCAD.cpp
+++ b/src/HeeksCAD.cpp
@@ -637,8 +637,8 @@ void HeeksCADapp::SetInputMode(CInputMode *new_mode){
        else{
                input_mode_object = m_select_mode;
        }
-       if(m_frame && m_frame->m_input_canvas)m_frame->RefreshInputCanvas();
-       if(m_frame && m_frame->m_options)m_frame->RefreshOptions();
+//     if(m_frame && m_frame->m_input_canvas)m_frame->RefreshInputCanvas();
+//     if(m_frame && m_frame->m_options)m_frame->RefreshOptions();
        if(m_graphics_text_mode != GraphicsTextModeNone)Repaint();
 }
 
```

neomilium, 2016-12-20:
@matael Does it work now under Arch ?

dunkelstern, 2017-01-15:
No sadly it does not.

Backtrace:
```
GNU gdb (GDB) 7.12
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from heekscad...(no debugging symbols found)...done.
(gdb) run
Starting program: /usr/bin/heekscad 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
05:02:49: Warning: Mismatch between the program and library build versions detected.
The library used 3.0 (wchar_t,compiler with C++ ABI 1009,wx containers,compatible with 2.8),
and your program used 3.0 (wchar_t,compiler with C++ ABI 1010,wx containers,compatible with 2.8).
Gtk-Message: (for origin information, set GTK_DEBUG): failed to retrieve property `gtk-primary-button-warps-slider' of type `gboolean' from rc file value "((GString*) 0xaf5840)" of type `gboolean'
05:02:49 AM: Debug: Adding duplicate image handler for 'PNG file'
Attempt to automatically load HeeksCNC: /usr/lib/libheekscnc.so
Resource: script_ops.xml found at: /usr/share/heekscnc/script_ops.xml
Resource: script_ops.xml found at: /usr/share/heekscnc/script_ops.xml

Program received signal SIGSEGV, Segmentation fault.
std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >::_M_assign (this=0xb73148, 
    __str=<error reading variable: Cannot access memory at address 0x8>)
    at /build/gcc/src/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.tcc:259
259	/build/gcc/src/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.tcc: No such file or directory.
(gdb) where
#0  0x00007ffff146a30e in std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >::_M_assign(std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> > const&) (this=0xb73148, __str=<error reading variable: Cannot access memory at address 0x8>)
    at /build/gcc/src/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.tcc:259
#1  0x0000000000675bdb in wxPGProperty::Init(wxString const&, wxString const&) ()
#2  0x0000000000675d33 in wxPGProperty::wxPGProperty(wxString const&, wxString const&) ()
#3  0x00000000006a5d16 in wxStringPropertyClass::wxStringPropertyClass(wxString const&, wxString const&, wxString const&) ()
#4  0x00000000006a5db1 in wxStringProperty(wxString const&, wxString const&, wxString const&) ()
#5  0x00000000005d9952 in CPropertiesCanvas::AddProperty(Property*, wxPGProperty*) ()
#6  0x00000000005bc43a in CInputModeCanvas::RefreshByRemovingAndAddingAll2() ()
#7  0x00000000005d8052 in CPropertiesCanvas::RefreshByRemovingAndAddingAll() ()
#8  0x0000000000569562 in HeeksCADapp::SetInputMode(CInputMode*) ()
#9  0x00000000005823af in HeeksCADapp::OnInit() ()
#10 0x00007ffff7a3a202 in wxEntry(int&, wchar_t**) () at /usr/lib/libwx_baseu-3.0.so.0
#11 0x00000000004ba052 in main ()
(gdb)
```

Tell me if i can do anything to debug.

dunkelstern, 2017-01-15:
Ok now this is strange. With the two lines commented as shown above it now starts every second time.

1. Run heekscad -> Works
2. Close
3. Run again -> Crashes
4. Run again -> Works

New backtrace:
```
(gdb) run
Starting program: /usr/bin/heekscad 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
17:00:40: Warning: Mismatch between the program and library build versions detected.
The library used 3.0 (wchar_t,compiler with C++ ABI 1009,wx containers,compatible with 2.8),
and your program used 3.0 (wchar_t,compiler with C++ ABI 1010,wx containers,compatible with 2.8).
Gtk-Message: (for origin information, set GTK_DEBUG): failed to retrieve property `gtk-primary-button-warps-slider' of type `gboolean' from rc file value "((GString*) 0xc14a40)" of type `gboolean'
05:00:40 PM: Debug: Adding duplicate image handler for 'PNG file'

Program received signal SIGSEGV, Segmentation fault.
std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >::_M_assign (this=0x1315138, 
    __str=<error reading variable: Cannot access memory at address 0x8>)
    at /build/gcc/src/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.tcc:259
259	/build/gcc/src/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.tcc: No such file or directory.
(gdb) where
#0  0x00007ffff121e30e in std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >::_M_assign(std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> > const&) (this=0x1315138, __str=<error reading variable: Cannot access memory at address 0x8>)
    at /build/gcc/src/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.tcc:259
#1  0x00000000006761fb in wxPGProperty::Init(wxString const&, wxString const&) ()
#2  0x0000000000676353 in wxPGProperty::wxPGProperty(wxString const&, wxString const&) ()
#3  0x0000000000676cbe in wxPGPropertyWithChildren::wxPGPropertyWithChildren(wxString const&, wxString const&) ()
#4  0x00000000006775e5 in wxParentProperty(wxString const&, wxString const&) ()
#5  0x00000000005da98a in CPropertiesCanvas::AddProperty(Property*, wxPGProperty*) ()
#6  0x00000000005ca7be in COptionsCanvas::RefreshByRemovingAndAddingAll2() ()
#7  0x00000000005d89e4 in CPropertiesCanvas::OnDraw(wxDC&) ()
#8  0x00007ffff75a4270 in wxAnyScrollHelperBase::HandleOnPaint(wxPaintEvent&) () at /usr/lib/libwx_gtk2u_core-3.0.so.0
#9  0x00007ffff75a6d8b in wxScrollHelperEvtHandler::ProcessEvent(wxEvent&) () at /usr/lib/libwx_gtk2u_core-3.0.so.0
#10 0x00007ffff7b2cd77 in wxEvtHandler::SafelyProcessEvent(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#11 0x00007ffff7357ce9 in wxWindow::GTKSendPaintEvents(_GdkRegion const*) () at /usr/lib/libwx_gtk2u_core-3.0.so.0
#12 0x00007ffff7358494 in  () at /usr/lib/libwx_gtk2u_core-3.0.so.0
#13 0x00007fffefce67ac in  () at /usr/lib/libgtk-x11-2.0.so.0
#14 0x00007fffeed10f75 in g_closure_invoke () at /usr/lib/libgobject-2.0.so.0
#15 0x00007fffeed22f82 in  () at /usr/lib/libgobject-2.0.so.0
#16 0x00007fffeed2b66f in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#17 0x00007fffeed2bfaf in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#18 0x00007fffefdfe8cc in  () at /usr/lib/libgtk-x11-2.0.so.0
#19 0x00007fffefce5592 in gtk_main_do_event () at /usr/lib/libgtk-x11-2.0.so.0
#20 0x00007fffef94329f in  () at /usr/lib/libgdk-x11-2.0.so.0
#21 0x00007fffef943245 in  () at /usr/lib/libgdk-x11-2.0.so.0
#22 0x00007fffef943245 in  () at /usr/lib/libgdk-x11-2.0.so.0
#23 0x00007fffef93fd71 in  () at /usr/lib/libgdk-x11-2.0.so.0
#24 0x00007fffef940688 in gdk_window_process_all_updates () at /usr/lib/libgdk-x11-2.0.so.0
#25 0x00007fffefc6bb61 in  () at /usr/lib/libgtk-x11-2.0.so.0
#26 0x00007fffef91f4b7 in  () at /usr/lib/libgdk-x11-2.0.so.0
#27 0x00007fffeea3843a in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
#28 0x00007fffeea387f0 in  () at /usr/lib/libglib-2.0.so.0
#29 0x00007fffeea38b12 in g_main_loop_run () at /usr/lib/libglib-2.0.so.0
#30 0x00007fffefce43a7 in gtk_main () at /usr/lib/libgtk-x11-2.0.so.0
#31 0x00007ffff7323165 in wxGUIEventLoop::DoRun() () at /usr/lib/libwx_gtk2u_core-3.0.so.0
#32 0x00007ffff79e8223 in wxEventLoopBase::Run() () at /usr/lib/libwx_baseu-3.0.so.0
#33 0x00007ffff79ad306 in wxAppConsoleBase::MainLoop() () at /usr/lib/libwx_baseu-3.0.so.0
#34 0x000000000058cfe9 in HeeksCADapp::OnRun() ()
#35 0x00007ffff7a3a1e0 in wxEntry(int&, wchar_t**) () at /usr/lib/libwx_baseu-3.0.so.0
#36 0x00000000004baa82 in main ()
(gdb) 
```

dunkelstern, 2017-01-15:
Ok "works" is some kind of misleading. The main window shows up. As soon as i create a sketch it crashes again:

```
#0  0x00007ffff121e30e in std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >::_M_assign(std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> > const&) (this=0x13d6988, __str=<error reading variable: Cannot access memory at address 0x8>)
    at /build/gcc/src/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.tcc:259
#1  0x00000000006761fb in wxPGProperty::Init(wxString const&, wxString const&) ()
#2  0x0000000000676353 in wxPGProperty::wxPGProperty(wxString const&, wxString const&) ()
#3  0x00000000006a6366 in wxStringPropertyClass::wxStringPropertyClass(wxString const&, wxString const&, wxString const&) ()
#4  0x00000000006a6401 in wxStringProperty(wxString const&, wxString const&, wxString const&) ()
#5  0x00000000005da522 in CPropertiesCanvas::AddProperty(Property*, wxPGProperty*) ()
#6  0x00000000005bd08a in CInputModeCanvas::RefreshByRemovingAndAddingAll2() ()
#7  0x00000000005d8c22 in CPropertiesCanvas::RefreshByRemovingAndAddingAll() ()
#8  0x000000000050382b in Drawing::OnMouse(wxMouseEvent&) ()
#9  0x0000000000517e1b in CViewport::ViewportOnMouse(wxMouseEvent&) ()
#10 0x0000000000517e84 in CGraphicsCanvas::OnMouse(wxMouseEvent&) ()
#11 0x00007ffff79a643e in wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const ()
    at /usr/lib/libwx_baseu-3.0.so.0
#12 0x00007ffff7b2ca77 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#13 0x00007ffff7b2cb6b in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) () at /usr/lib/libwx_baseu-3.0.so.0
#14 0x00007ffff7b2cf28 in wxEvtHandler::TryHereOnly(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#15 0x00007ffff7b2cfb3 in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#16 0x00007ffff7b2d015 in wxEvtHandler::ProcessEvent(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#17 0x00007ffff7b2cd77 in wxEvtHandler::SafelyProcessEvent(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#18 0x00007ffff73520c0 in  () at /usr/lib/libwx_gtk2u_core-3.0.so.0
#19 0x00007fffefce67ac in  () at /usr/lib/libgtk-x11-2.0.so.0
#20 0x00007fffeed10f75 in g_closure_invoke () at /usr/lib/libgobject-2.0.so.0
#21 0x00007fffeed22f82 in  () at /usr/lib/libgobject-2.0.so.0
#22 0x00007fffeed2b66f in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#23 0x00007fffeed2bfaf in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#24 0x00007fffefdfe8cc in  () at /usr/lib/libgtk-x11-2.0.so.0
#25 0x00007fffefce4f74 in gtk_propagate_event () at /usr/lib/libgtk-x11-2.0.so.0
#26 0x00007fffefce532b in gtk_main_do_event () at /usr/lib/libgtk-x11-2.0.so.0
#27 0x00007fffef95c46c in  () at /usr/lib/libgdk-x11-2.0.so.0
#28 0x00007fffeea38587 in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
#29 0x00007fffeea387f0 in  () at /usr/lib/libglib-2.0.so.0
#30 0x00007fffeea38b12 in g_main_loop_run () at /usr/lib/libglib-2.0.so.0
#31 0x00007fffefce43a7 in gtk_main () at /usr/lib/libgtk-x11-2.0.so.0
#32 0x00007ffff7323165 in wxGUIEventLoop::DoRun() () at /usr/lib/libwx_gtk2u_core-3.0.so.0
#33 0x00007ffff79e8223 in wxEventLoopBase::Run() () at /usr/lib/libwx_baseu-3.0.so.0
#34 0x00007ffff79ad306 in wxAppConsoleBase::MainLoop() () at /usr/lib/libwx_baseu-3.0.so.0
#35 0x000000000058cfe9 in HeeksCADapp::OnRun() ()
#36 0x00007ffff7a3a1e0 in wxEntry(int&, wchar_t**) () at /usr/lib/libwx_baseu-3.0.so.0
#37 0x00000000004baa82 in main ()
```

jb36, 2017-01-20:
I just tested again and I needed to specify the C++98 standard to get it working. I'm not sure that this is the best way to do it, but here are the changes that I made.

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index aa5892e..c8c7a5d 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -8,6 +8,9 @@ find_package( wxWidgets REQUIRED COMPONENTS base core gl )
 #set additionnal CMake module path
 set( CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${CMAKE_SOURCE_DIR}/cmake/Modules/")
 
+#set( CMAKE_BUILD_TYPE Debug )
+set( CMAKE_CXX_FLAGS "-std=c++98" )
+
 #find OCE or OpenCASCADE
 set( CASCADE_LIBS TKernel )
 if( UNIX )
diff --git a/src/CMakeLists.txt b/src/CMakeLists.txt
index 7978096..9998279 100644
--- a/src/CMakeLists.txt
+++ b/src/CMakeLists.txt
@@ -3,6 +3,9 @@ cmake_minimum_required( VERSION 2.6 )
 
 add_definitions ( -Wall -DUSE_GENERIC_TREECTRL -DHEEKSCAD -DUNICODE -DTIXML_USE_STL )
 
+#set( CMAKE_BUILD_TYPE Debug )
+set( CMAKE_CXX_FLAGS "-std=c++98" )
+
 #find OCE or OpenCASCADE
 set( CASCADE_LIBS TKBRep TKIGES TKSTEP TKXSBase TKBool TKernel TKFillet TKG2d TKG3d TKGeomAlgo TKGeomBase TKMath TKOffset TKPrim TKTopAlgo TKV3d TKMesh TKBO )
 set( CASCADE_NOTUSED_LIBS TKPShape TKShHealing TKLCAF TKFeat TKXCAF TKHLR TKPLCAF TKSTL TKSTEP209 TKAdvTools TKCAF TKV2d TKVRML TKXCAFSchema TKXDESTEP TKSTEPAttr FWOSPlugin TKSTEPBase TKService TKShapeSchema TKCDF )
```

greyltc, 2017-02-15:
Strange, I was having these segfaults on Arch Linux too and rather than
`set( CMAKE_CXX_FLAGS "-std=c++98" )` as above, I had to do `set( CMAKE_CXX_FLAGS "-std=c++11" )` to get things working.

neomilium, 2017-03-26:
Maybe it related to flags that have been used to compile wxWidgets.
@greyltc , @jb36 could you check this ?

jb36, 2017-03-27:
Now Arch has moved on to wxWidgets version 3. That does need the c++11 flag to build. I have tried it only briefly, but was able to create a simple sketch.

The version that I am actively using was compiled with wxWidgets version 2.8. Doing this with both 3.0 and 2.8 installed means running:
```
cmake -DCMAKE_INSTALL_PREFIX=/usr -D wxWidgets_CONFIG_EXECUTABLE:FILEPATH=/usr/bin/wx-config-2.8 .
```

For the flags used to compile wxWidgets, I used the PKGBUILD from here: [https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=wxgtk2.8](url)

```
$ pacman -Q wxgtk2.8
wxgtk2.8 2.8.12.1-6
```

l29ah, 2017-05-30:
Got the same problem on Gentoo on wxWidgets-3.0.

l29ah, 2017-08-27:
Ok, the problem boils down to wxWidgets and heekscad being built w/ different C++ compiler{, version}s.

Matael, 2017-09-03:
I got a working binary using this PKGBUILD on an up-to-date Arch for heekscad:

```
# Maintainer: Grey Christoforo <first name at last name dot net>
pkgname=heekscad-git
pkgver=1451.028ca9ac
pkgrel=1
pkgdesc="HeeksCNC is an open source 3D CAD/CAM software"
arch=('x86_64')
url="http://heeks.net"
license=('custom:BSD3')
depends=('python2' 'oce' 'wxgtk2' 'libarea-git')
optdepends=('heekscnc-git')
makedepends=('cmake')
provides=('heekscad')
conflicts=('heekscad')
source=('git://github.com/Heeks/heekscad.git')
md5sums=('SKIP')

pkgver() {
  cd "${srcdir}/heekscad"
  echo $(git rev-list --count HEAD).$(git rev-parse --short HEAD)
}

prepare() {
  # fix the hardcoded calls to python
  sed -i 's,#find OCE or OpenCASCADE,set( CMAKE_CXX_FLAGS "-std=c++11" )\n#find OCE or OpenCASCADE,g' "${srcdir}/heekscad/CMakeLists.txt"
  sed -i 's,#find OCE or OpenCASCADE,set( CMAKE_CXX_FLAGS "-std=c++11" )\n#find OCE or OpenCASCADE,g' "${srcdir}/heekscad/src/CMakeLists.txt"
}

build() {
  msg "Starting build..."
  cd "${srcdir}/heekscad"
  mkdir -p build
  cd build
  export OCE_DIR="/opt/oce/lib/oce-0.18"
  cmake -DwxWidgets_CONFIG_EXECUTABLE="/usr/bin/wx-config" -DPYTHON_EXECUTABLE="/usr/bin/python2" -DPYTHON_LIBRARY="/usr/lib/libpython2.7.so" -DPYTHON_INCLUDE_DIR="/usr/include/python2.7" -DCMAKE_INSTALL_PREFIX=/usr ..
}

package() {
  cd "$srcdir/heekscad/build"
  make DESTDIR="$pkgdir/" install

  mkdir -p "${pkgdir}/usr/share/licenses"
  cp "${srcdir}/heekscad/COPYING" "${pkgdir}/usr/share/licenses/$pkgname"
}
```

and for Heekscnc:

```
# Maintainer: Grey Christoforo <first name at last name dot net>
pkgname=heekscnc-git
pkgver=1143.8c70474
pkgrel=7
pkgdesc="CAM add-on for HeeksCAD"
arch=('x86_64')
url="http://heeks.net"
license=('custom:BSD3')
depends=('heekscad-git')
makedepends=('cmake')
source=('git://github.com/Heeks/heekscnc.git' grbl.py)
provides=('heekscnc')
conflicts=('heekscnc')
md5sums=('SKIP'
         '05033d96056948b45059df301216a42d')

pkgver() {
  cd "${srcdir}/heekscnc"
  #echo $(git describe --long | sed 's/\([^-]*-g\)/r\1/;s/-/./g')
  echo $(git rev-list --count HEAD).$(git rev-parse --short HEAD)
}

prepare() {
  cd "${srcdir}/heekscnc"

  # add grbl support
  cp ../grbl.py nc/.
  echo '<Machine post="grbl" reader="iso_read" suffix=".ngc" description="GRBL"/>' >> nc/machines.xml

  # fix hardcoded call to python
  sed -i 's,Execute(wxString(_T("python \\"")) + path + wxString(_T("backplot\.py\\" \\"")) + m_program->m_machine\.reader + wxString(_T("\\" \\"")) + m_filename + wxString(_T("\\"")) );,Execute(wxString(_T("python2 \\"")) + path + wxString(_T("backplot\.py\\" \\"")) + m_program->m_machine\.reader + wxString(_T("\\" \\"")) + m_filename + wxString(_T("\\"")) );,g' src/PythonStuff.cpp
  sed -i 's,wxString post_path = wxString(_T("python ")) + path.GetFullPath();,wxString post_path = wxString(_T("python2 ")) + path.GetFullPath();,g' src/PythonStuff.cpp
}

build() {
  msg "Starting build..."
  cd "${srcdir}/heekscnc"
  mkdir -p build
  cd build
  export OCE_DIR="/opt/oce/lib/oce-0.18"
  cmake -DwxWidgets_CONFIG_EXECUTABLE="/usr/bin/wx-config" -DPYTHON_EXECUTABLE="/usr/bin/python2" -DPYTHON_LIBRARY="/usr/lib/libpython2.7.so" -DPYTHON_INCLUDE_DIR="/usr/include/python2.7" -DCMAKE_INSTALL_PREFIX=/usr ..
}

package() {
  cd "$srcdir/heekscnc/build"
  make DESTDIR="$pkgdir/" install

  mkdir -p "${pkgdir}/usr/share/licenses"
  cp "${srcdir}/heekscnc/COPYING" "${pkgdir}/usr/share/licenses/$pkgname"
}
```

I notified Christoforo (arch pkg's maintainer) on AUR.

layus, 2018-09-18:
I have no idea how you managed to get something working. Using swig somehow I guess. The error arises in
https://github.com/Heeks/heekscad/blob/master/src/propgrid.cpp#L714-L727 when the if test is bypassed by the compiler (a reference is assumed never to be null).
But somehow this line manages to generate null references https://github.com/Heeks/heekscad/blob/master/src/propgrid.h#L424

Looking at the code, it seems that with SWIG and __WXPYTHON__, we can avoid most of these issues.

@neomilium, @danheeks Any idea on the best way to get SWIG defined ? Or to compile with swig ?