Heeks/heekscad #412: Tolerance settings doesn't modify HeeksCNC output program

lpgasparotto, 2016-09-07:
I'm using the Ubuntu PPA version of HeeksCAD/CNC. My OS is KDE Neon which is based on Ubuntu 16.04. I changed these parameters for getting a more accurate output:
- STL tolerance
- Surface tolerance
- Profile spline deviation
- Pocket spline deviation

But I always get the same number of lines. I used even extreme number.I also tried changing these parameters and then restart HeeksCAD but there was no changes. The piece on which I'm working is a complex one, with lots of arcs.
[image]

I'm trying to get smaller segments (like the one showed in the image), less distance between start/end points.

Is it normal the tolerance/resolution be fixed?

lpgasparotto, 2016-09-07:
Maybe this is caused by the size of /tmp/surface1.stl which is always the same, even setting extreme parameters.

lpgasparotto, 2016-09-11:
Definitely, it's not related to surface1.stl. I tested it with native high resolution STL files which made this file to be bigger but I always get the same code. It seems like something is fixed and doesn't allow to modify the overall tolerance.

neomilium, 2016-12-20:
Please use appropriated issue tracker : HeeksCNC

closed 2016-12-20 by neomilium