Heeks/heekscad #414: Build on Debian GNU/Linux 7.11 (wheezy) Linux CNC installation

callahanp, 2016-12-17:
sudo apt-get install git build-essential debhelper cmake libboost-python-dev python-dev
git clone https://github.com/Heeks/libarea.git
cd libarea
dpkg-buildpackage -b -us -uc
cd ../
sudo dpkg -i libarea*.deb python-area*.deb
sudo apt-get install liboce-visualization-dev libwxgtk2.8-dev libgtkglext1-dev python-dev
git clone https://github.com/Heeks/heekscad.gt
cd heekscad
dpkg-buildpackage -b -us -uc
dpkg-buildpackage -b -us -uc
dpkg-buildpackage: source package heekscad
dpkg-buildpackage: source version 1.1.0-1
dpkg-buildpackage: source changed by Romuald Conty <neomilium@gmail.com>
dpkg-buildpackage: host architecture i386
 dpkg-source --before-build heekscad
 fakeroot debian/rules clean
dh clean 
   dh_testdir
   dh_auto_clean
   dh_clean
 debian/rules build
dh build 
   dh_testdir
   dh_auto_configure
-- The C compiler identification is GNU 4.7.2
-- The CXX compiler identification is GNU 4.7.2
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Found wxWidgets: TRUE  
CMake Error at /usr/lib/oce-0.9.1/OCE-libraries-release.cmake:732 (MESSAGE):
  The imported target "TKCAF" references the file

     "/usr/lib/libTKCAF.so.2.0.0"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/lib/oce-0.9.1/OCE-libraries-release.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /usr/lib/oce-0.9.1/OCE-libraries.cmake:214 (INCLUDE)
  /usr/lib/oce-0.9.1/OCEConfig.cmake:21 (INCLUDE)
  CMakeLists.txt:18 (find_package)

-- Configuring incomplete, errors occurred!
dh_auto_configure: cmake .. -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_VERBOSE_MAKEFILE=ON returned exit code 1
make: *** [build] Error 2
dpkg-buildpackage: error: debian/rules build gave error exit status 2

while LIBTKCAF appears to be missing, but there is a LIBTKCDF:

/usr/lib/libTKCDF.so
/usr/lib/libTKCDF.so.2
/usr/lib/libTKCDF.so.2.0.0

neomilium, 2016-12-20:
Output message tells you you miss libTKCAF which is a part of `liboce-ocaf-dev` AFAICS on debian packages.

Could you try to install it then restart build ?

callahanp, 2016-12-20:
On Mon, 19 Dec 2016 18:03:56 -0800
Romuald Conty <notifications@github.com> wrote:

> Output message tells you you miss libTKCAF which is a part of
> `liboce-ocaf-dev` AFAICS on debian packages.
> 
> Could you try to install it then restart build ?
> 
It was already installed.

sudo apt-get install liboce-ocaf-dev
[sudo] password for pac1: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
liboce-ocaf-dev is already the newest version (0.17.2-1).
0 upgraded, 0 newly installed, 0 to remove and 15 not upgraded.

neomilium, 2016-12-20:
Where does your package comes from ?
You have version 0.17.2 but you said you are under Wheezy and this distribution have 0.9.1.
https://packages.debian.org/wheezy/liboce-ocaf-dev

callahanp, 2016-12-20:
from git master.

On Tue, 20 Dec 2016 02:30:01 -0800
Romuald Conty <notifications@github.com> wrote:

> Where does your package comes from ?
> You have version 0.17.2 but you said you are under Wheezy and this
> distribution have 0.9.1.
> https://packages.debian.org/wheezy/liboce-ocaf-dev
>

neomilium, 2016-12-20:
So, check your OCE installation : you miss libTKCAF

closed 2016-12-20 by neomilium