Heeks/heekscad #415: Error when invoking the G0 machining tool

daveseff, 2017-01-27:
11:18:35 AM: Traceback (most recent call last):
  File "/tmp/post.py", line 6, in <module>
    import area
ImportError: /usr/lib/libboost_python.so.1.63.0: undefined symbol: PyClass_Type

11:18:35 AM: python: can't open file '/usr/lib/heekscnc/backplot.py': [Errno 2] No such file or directory

11:18:35 AM: can't open file '/tmp/backplot.xml' (error 2: No such file or directory)

daveseff, 2017-01-27:
This looks like an issue with libarea and not heekscad. Also This is more related to heekscnc than heekscad.

closed 2017-01-27 by daveseff

juup, 2018-11-16:
Hi daveseff,

did you find a solution?
Heekscad worked fine on my Arch system for a while. Now I get the same error:

```
09:53:08: Traceback (most recent call last):
  File "/tmp/post.py", line 6, in <module>
    import area
ImportError: libboost_python27.so.1.67.0: cannot open shared object file: No such file or directory

09:53:08: Traceback (most recent call last):
  File "/usr/lib/heekscnc/backplot.py", line 8, in <module>
    machine_module = __import__('nc.' + reader, fromlist = ['dummy'])
  File "/usr/lib/heekscnc/nc/iso_read.py", line 8, in <module>
    import nc_read as nc
  File "/usr/lib/heekscnc/nc/nc_read.py", line 7, in <module>
    import area
ImportError: libboost_python27.so.1.67.0: cannot open shared object file: No such file or directory

09:53:08: Kann Datei '/tmp/backplot.xml' nicht öffnen (Fehler 2: Datei oder Verzeichnis nicht gefunden)
```

d3tic, 2020-07-10:
I have just upgraded to 20.04 ubuntu and I get a similar error.  Everything is fine on 18.04 but mu 18.04 is on a flaky disk and takes a long time to boot.

neomilium, 2020-07-10:
@d3tic how heekscad was installed on your system?

d3tic, 2020-07-11:
via the ppa method.  I have done a bit more investigations and I get a
slightly different error using python 2(.7) or the default for fossa 3(.8).

The problem seems to be with libarea as I can't even import it from the
command line.
I tried to install libarea from deb packages which would only work for 2.7
but I still get the error
and can't import it.  I also tried just copying the lib files to the right
locations.

I used to be a suse system admin many years ago but have only recently
started using Linux on the desktop. Been using
EMC/linuxcnc for over 10 years.

I think my ubuntu 20.04 might be slightly stuffed as it was an upgrade from
19.whatever kubuntu. I still see kubuntu flash
screens and I had duplicates of some apps vlc, freecad and others also the
update disabled theheeks ppa sources.

I can still use heekscad as I have a dual boot to 18.04 but the disk it is
on has bad sectors and can take over 5 minutes
to start and everything is slow and it may die completely soon.

d3tic, 2020-07-11:
OK i got it to work with a dirty hack.

1. set python to version 2

awg@lappy-TV:/usr/lib$ sudo update-alternatives --config python
There are 2 choices for the alternative python (providing /usr/bin/python).

  Selection    Path              Priority   Status
------------------------------------------------------------
  0            /usr/bin/python3   2         auto mode
* 1            /usr/bin/python2   1         manual mode
  2            /usr/bin/python3   2         manual mode

2. copied old libboost from 18.04 and linked it to new ver

lrwxrwxrwx 1 root root      30 Jul 11 15:58 libboost_python38.so ->
libboost_python-py27.so.1.65.1
lrwxrwxrwx 1 root root      30 Jul 11 16:02 libboost_python38.so.1.71.0 ->
libboost_python-py27.so.1.65.1

The first line didn't change anything so renamed the file with .bak and
made a link with the original name.

d3tic, 2020-07-28:
I think the PPA keeps changing the links I made back so I did a disk wipe
and fresh install of 20.04 and I still get the same error.
Do I need some python dev files....

I might have to reinstall windows to use my paid for version this is
getting so bad.