Heeks/heekscad #416: HeeksCAD crash changing any checkable propertie in Lubuntu 16.04 64bits

Arcachofo, 2017-03-27:
HeeksCAD crash changing any checkable propertie, for example changing visiblity, calculate volume of sphere, use zig-zag, or any other checkable item.

Tried stable neomilium ppa and built from sources.
Tried only HeeksCAD without HeeksCNC and the problem is still there.

ASSERT INFO:
../src/gtk/window.cpp(4096): assert "Assert failure" failed in GTKSendPaintEvents(): unsupported background style

BACKTRACE:
[1] wxWindow::GTKSendPaintEvents(_GdkRegion const*)
[2] g_closure_invoke
[3] g_signal_emit_valist
[4] g_signal_emit
[5] gtk_main_do_event
[6] gdk_window_process_all_updates
[7] g_main_context_dispatch
[8] g_main_loop_run
[9] gtk_main
[10] wxGUIEventLoop::DoRun()
[11] wxEventLoopBase::Run()
[12] wxAppConsoleBase::MainLoop()
[13] HeeksCADapp::OnRun()
[14] wxEntry(int&, wchar_t**)
[15] main
[16] __libc_start_main
[17] _start

Arcachofo, 2017-03-27:
I solved the issue this way:

heekscad/src/propgrid.cpp
wxSimpleCheckBox constructor, line 3062

Comment out line:
//SetBackgroundStyle( wxBG_STYLE_COLOUR );

neomilium, 2017-03-28:
Thanks.

Note : PPA should be updated in few days.

closed 2017-03-28 by neomilium

neomilium, 2017-03-28:
PPA is now up-to-date (devel and stable)

commit 028ca9ac (2017-03-28): Fix "GTKSendPaintEvents(): unsupported background style"

This commit fixes #416.
Thanks to Arcachofo.