Heeks/heekscad #420: Pocket gcode path overflow

woodm528, 2018-10-14:
Hi;
I am seeing a pocket gcode generation issue with a simple 2d plate. I have attached a couple screen shots and can provide the heeks file. I see this on heekscad version 1.3.0 (runing on ubuntu 18.04) as well as heekscad 1.1.0 (ubuntu 16.04). 
Not seeing any recent activity on here, but I figure id post any way. 
Also I was trying to pull the source but having trouble setting up dependencies referenced in README.md openscade etc, Not sure where to get additional build detail. Any help would be greatly appreciated

Thanks
Mike    

[image]

[image]

woodm528, 2018-10-14:
Oh okay - so needed to zip the heeks file to get it to attach, it is attached here
Thanks
Mike 

[z_ball_screw_mount_plate.heeks.zip](https://github.com/Heeks/heekscad/files/2476073/z_ball_screw_mount_plate.heeks.zip)

philtipping, 2018-12-13:
Mike, just tried your zip file with v1.4.0 on lubuntu 18.04. Clicked G0 and code generated with no errors.
Phil.

woodm528, 2018-12-16:
Hi Phil; 
Thanks for taking a look at this. So yours works and mine doesn't. Is there something I can do to check my install. Maybe make sure I have all the right versions of heeks and dependent libraries, etc.

Thanks
Mike

woodm528, 2018-12-16:
Looks like your a version ahead of mine. Not seen this sight either. Looks like i'm gonna be updating my app when I get the time. I post the result after.

Fyi;  My notes indicate I had installed via apt-get following this sight 
https://github.com/Heeks/heekscnc/wiki/Getting-HeeksCNC-for-Linux
sudo add-apt-repository ppa:neomilium/heekscnc-devel
sudo apt-get update
sudo apt-get install heekscnc

Thanks
Mike

philtipping, 2018-12-16:
Ok Mike - sounds like you're better organised than me! I've got references to 'ppa.launchpad.net/neomilium/heekscnc-devel/ubuntu bionic InRelease' when I do a  sudo apt-get update, so this suggests I must have followed the ubuntu 18.04 instructions. I mainly use the Windows 1.4 version as my target Mach3 runs on an old Windows XP computer, but I have the lubuntu version on a laptop for armchair 'learning' - only made my first 'cut' this week :)  Let me know how you get on.