Heeks/heekscad #422: Cannot open /tmp/backplot.xml

juup, 2019-10-24:
I already comment to a similar issue #415, but because it is closed there was no reply.

Heekscad worked fine on my Arch system for a while. Now I get this error:

```
#09:53:08: Traceback (most recent call last):
  File "/tmp/post.py", line 6, in <module>
    import area
ImportError: libboost_python27.so.1.67.0: cannot open shared object file: No such file or directory

09:53:08: Traceback (most recent call last):
  File "/usr/lib/heekscnc/backplot.py", line 8, in <module>
    machine_module = __import__('nc.' + reader, fromlist = ['dummy'])
  File "/usr/lib/heekscnc/nc/iso_read.py", line 8, in <module>
    import nc_read as nc
  File "/usr/lib/heekscnc/nc/nc_read.py", line 7, in <module>
    import area
ImportError: libboost_python27.so.1.67.0: cannot open shared object file: No such file or directory

09:53:08: Kann Datei '/tmp/backplot.xml' nicht öffnen (Fehler 2: Datei oder Verzeichnis nicht gefunden)
```

neomilium, 2019-10-24:
Your problem is related to `ImportError: libboost_python27.so.1.67.0: cannot open shared object file: No such file or directory` while importing `area` python module, as described by backtrace.

So you should have a look why `boost` python module, for python 2.7, is missing.
Double-check:
* if you the `boost` python package is installed?
* if the used python version for your system is 2.7
* if the `area` python module does need to be recompile against an updated dependency

Note: As you problem is _only_ related to the missing `libboost_python27.so.1.67.0` file, you should re-title the issue.

neomilium, 2019-10-24:
And for the record, it not related to #415

closed 2019-10-24 by neomilium

juup, 2019-10-25:
Thanks neomilium for the quick reply.
On my system are python 2.7 and 3.7 installed.
When looking for `boos`:
```
pacman -Ss libboost_python27
extra/boost-libs 1.71.0-2 [Installiert]
```
Sorry for the beginner question: do I have to downgrade boost-libs to 1.67?