Heeks/heekscad #426: Change tool comment confuses some machines and simulators

zio-pietro, 2019-11-13:
Gcode postprocessors prepend comment to change tool command in form
`(tool change to <tool name> (<tool type>) )`
see **nested** parenthesis!
But according to infos found in net (for ex. cnczone.com)

> Printable characters and white space inside parentheses is a comment. A left parenthesis always starts a comment. The comment ends at _the first_ right parenthesis found thereafter.
> Once a left parenthesis is placed on a line, a matching right parenthesis must appear before the end of the line. [_multiline comments of type 'just a pair of brackets for all lines' aren't allowed_]
> Comments _**may not** be nested_; it is an error if a left parenthesis is found after the start of a comment and before the end of the comment.

(note and emphasis mine)

Not tried on my CNC mill yet, but simulator _CAMotics_ refuse load that Gcode with errormsg
```
ERROR: Invalid character: ')'
ERROR: At: <gcode filename> <line>:<location of second right parenthesis>

```
Consistent with the cnczone comment, CAMotics parser closes comment at first occurrence of right parenthesis and a second one is considered as invalid character.

Q&D workaround: open gcode file with text editor and cancel guilty parenthesis ;)
Doing so every time is boring, so I suggest change comment form to

`(tool change to <tool name> - <tool type>)`

Just my 5 cents.

zio-pietro, 2019-11-13:
Oh, wrong place, it's heekscnc issue, sorry for it. :(
Can it be moved there or I must repost it?

mehw, 2019-11-14:
> Can it be moved there or I must repost it?

I think that the owner of the repos can move an issue from one to another: [Transferring an issue to another repository](https://help.github.com/en/github/managing-your-work-on-github/transferring-an-issue-to-another-repository)

Good luck!