Heeks/heekscad #428: Linking error: SetPickingColor

tarnh, 2019-12-06:
Since I haven't paid for your software I don't demand any help. This is just for documentation.
compiler = gcc 9.2

```
[100%] Linking CXX executable ../bin/heekscad
/usr/bin/ld: CMakeFiles/heekscad.dir/HeeksCAD.cpp.o: in function `HeeksCADapp::glCommands(bool, bool, bool)':
HeeksCAD.cpp:(.text+0xf68b): undefined reference to `SetPickingColor(unsigned int)'
/usr/bin/ld: HeeksCAD.cpp:(.text+0xf73c): undefined reference to `SetPickingColor(unsigned int)'
/usr/bin/ld: CMakeFiles/heekscad.dir/MarkedList.cpp.o: in function `MarkedList::render_move_grips(bool, bool)':
MarkedList.cpp:(.text+0x938): undefined reference to `SetPickingColor(unsigned int)'
/usr/bin/ld: CMakeFiles/heekscad.dir/MarkedList.cpp.o: in function `MarkedList::ObjectsInWindow(wxRect, MarkedObject*, bool)':
MarkedList.cpp:(.text+0xd92): undefined reference to `GetPickingName(unsigned char, unsigned char, unsigned char)'
/usr/bin/ld: MarkedList.cpp:(.text+0xe46): undefined reference to `GetPickingName(unsigned char, unsigned char, unsigned char)'
make[2]: *** [src/CMakeFiles/heekscad.dir/build.make:1543: bin/heekscad-1.0.0] Error 1
make[1]: *** [CMakeFiles/Makefile2:238: src/CMakeFiles/heekscad.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

bianchifan, 2019-12-26:
Same here...
Dell Studio 1535 4GB, Ubuntu 19/10, GCC 9.2.1.

majo@majo-Studio-1535:~/git/heekscad$ dpkg-buildpackage -b -us -uc
dpkg-buildpackage: Information: Quellpaket heekscad
dpkg-buildpackage: Information: Quellversion 1.1.0-1
dpkg-buildpackage: Information: Quelldistribution unstable
dpkg-buildpackage: Information: Quelle geändert durch Romuald Conty <neomilium@gmail.com>
dpkg-buildpackage: Information: Host-Architektur amd64
 dpkg-source --before-build .
 fakeroot debian/rules clean
dh clean 
dh: Compatibility levels before 9 are deprecated (level 7 in use)
   dh_clean
dh_clean: Compatibility levels before 9 are deprecated (level 7 in use)
 debian/rules build
dh build 
dh: Compatibility levels before 9 are deprecated (level 7 in use)
   dh_update_autotools_config
   dh_auto_configure
dh_auto_configure: Compatibility levels before 9 are deprecated (level 7 in use)
	cd obj-x86_64-linux-gnu && cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON "-GUnix Makefiles" -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_INSTALL_LIBDIR=lib/x86_64-linux-gnu ..
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
.
.
.
/usr/include/c++/9/bits/unique_ptr.h:53:28: note: declared here
   53 |   template<typename> class auto_ptr;
      |                            ^~~~~~~~
[100%] Linking CXX executable ../bin/heekscad
cd /home/majo/git/heekscad/obj-x86_64-linux-gnu/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/heekscad.dir/link.txt --verbose=1
/usr/bin/c++   -pthread -pthread  -rdynamic CMakeFiles/heekscad.dir/AboutBox.cpp.o CMakeFiles/heekscad.dir/AutoSave.cpp.o CMakeFiles/heekscad.dir/BezierCurve.cpp.o CMakeFiles/heekscad.dir/Cone.cpp.o CMakeFiles/heekscad.dir/ConversionTools.cpp.o CMakeFiles/heekscad.dir/CoordinateSystem.cpp.o CMakeFiles/heekscad.dir/CorrelationTool.cpp.o CMakeFiles/heekscad.dir/Cuboid.cpp.o CMakeFiles/heekscad.dir/CxfFont.cpp.o CMakeFiles/heekscad.dir/Cylinder.cpp.o CMakeFiles/heekscad.dir/DigitizeMode.cpp.o CMakeFiles/heekscad.dir/DigitizedPoint.cpp.o CMakeFiles/heekscad.dir/DimensionDrawing.cpp.o CMakeFiles/heekscad.dir/Drawing.cpp.o CMakeFiles/heekscad.dir/Edge.cpp.o CMakeFiles/heekscad.dir/EndedObject.cpp.o CMakeFiles/heekscad.dir/ExitMainLoop.cpp.o CMakeFiles/heekscad.dir/Face.cpp.o CMakeFiles/heekscad.dir/FaceTools.cpp.o CMakeFiles/heekscad.dir/FastCurves.cpp.o CMakeFiles/heekscad.dir/GLList.cpp.o CMakeFiles/heekscad.dir/GraphicsCanvas.cpp.o CMakeFiles/heekscad.dir/Grid.cpp.o CMakeFiles/heekscad.dir/Gripper.cpp.o CMakeFiles/heekscad.dir/GripperSelTransform.cpp.o CMakeFiles/heekscad.dir/Group.cpp.o CMakeFiles/heekscad.dir/HAngularDimension.cpp.o CMakeFiles/heekscad.dir/HArc.cpp.o CMakeFiles/heekscad.dir/HArea.cpp.o CMakeFiles/heekscad.dir/HCircle.cpp.o CMakeFiles/heekscad.dir/HDimension.cpp.o CMakeFiles/heekscad.dir/HDxf.cpp.o CMakeFiles/heekscad.dir/HEllipse.cpp.o CMakeFiles/heekscad.dir/HGear.cpp.o CMakeFiles/heekscad.dir/HILine.cpp.o CMakeFiles/heekscad.dir/HImage.cpp.o CMakeFiles/heekscad.dir/HLine.cpp.o CMakeFiles/heekscad.dir/HPoint.cpp.o CMakeFiles/heekscad.dir/HSpline.cpp.o CMakeFiles/heekscad.dir/HText.cpp.o CMakeFiles/heekscad.dir/HXml.cpp.o CMakeFiles/heekscad.dir/HeeksCAD.cpp.o CMakeFiles/heekscad.dir/HeeksFrame.cpp.o CMakeFiles/heekscad.dir/HeeksPrintout.cpp.o CMakeFiles/heekscad.dir/History.cpp.o CMakeFiles/heekscad.dir/Input.cpp.o CMakeFiles/heekscad.dir/InputModeCanvas.cpp.o CMakeFiles/heekscad.dir/LineArcDrawing.cpp.o CMakeFiles/heekscad.dir/Loop.cpp.o CMakeFiles/heekscad.dir/MagDragWindow.cpp.o CMakeFiles/heekscad.dir/MarkedList.cpp.o CMakeFiles/heekscad.dir/ObjPropsCanvas.cpp.o CMakeFiles/heekscad.dir/OptionsCanvas.cpp.o CMakeFiles/heekscad.dir/OrientationModifier.cpp.o CMakeFiles/heekscad.dir/Plugins.cpp.o CMakeFiles/heekscad.dir/PointDrawing.cpp.o CMakeFiles/heekscad.dir/PointOrWindow.cpp.o CMakeFiles/heekscad.dir/PropertiesCanvas.cpp.o CMakeFiles/heekscad.dir/PropertyChange.cpp.o CMakeFiles/heekscad.dir/PropertyTrsf.cpp.o CMakeFiles/heekscad.dir/RS274X.cpp.o CMakeFiles/heekscad.dir/RegularShapesDrawing.cpp.o CMakeFiles/heekscad.dir/RemoveOrAddTool.cpp.o CMakeFiles/heekscad.dir/RuledSurface.cpp.o CMakeFiles/heekscad.dir/Ruler.cpp.o CMakeFiles/heekscad.dir/Sectioning.cpp.o CMakeFiles/heekscad.dir/SelectMode.cpp.o CMakeFiles/heekscad.dir/Shape.cpp.o CMakeFiles/heekscad.dir/ShapeData.cpp.o CMakeFiles/heekscad.dir/ShapeTools.cpp.o CMakeFiles/heekscad.dir/Sketch.cpp.o CMakeFiles/heekscad.dir/Solid.cpp.o CMakeFiles/heekscad.dir/SolidTools.cpp.o CMakeFiles/heekscad.dir/Sphere.cpp.o CMakeFiles/heekscad.dir/StlSolid.cpp.o CMakeFiles/heekscad.dir/StretchTool.cpp.o CMakeFiles/heekscad.dir/TransformTool.cpp.o CMakeFiles/heekscad.dir/TransformTools.cpp.o CMakeFiles/heekscad.dir/TreeCanvas.cpp.o CMakeFiles/heekscad.dir/Vertex.cpp.o CMakeFiles/heekscad.dir/ViewPanning.cpp.o CMakeFiles/heekscad.dir/V [... cut ...]
make[3]: Verzeichnis „/home/majo/git/heekscad/obj-x86_64-linux-gnu“ wird verlassen
make[2]: *** [CMakeFiles/Makefile2:259: src/CMakeFiles/heekscad.dir/all] Fehler 2
make[2]: Verzeichnis „/home/majo/git/heekscad/obj-x86_64-linux-gnu“ wird verlassen
make[1]: *** [Makefile:133: all] Fehler 2
make[1]: Verzeichnis „/home/majo/git/heekscad/obj-x86_64-linux-gnu“ wird verlassen
dh_auto_build: cd obj-x86_64-linux-gnu && make -j1 returned exit code 2
make: *** [debian/rules:8: build] Fehler 255
dpkg-buildpackage: Fehler: Unterprozess debian/rules build lieferte Exitstatus 2

dziedzic99, 2020-02-11:
I confirm this occurs on my Manjaro as well.

> [100%] Linking CXX executable ../bin/heekscad
/usr/bin/ld: CMakeFiles/heekscad.dir/HeeksCAD.cpp.o: in function `HeeksCADapp::glCommands(bool, bool, bool)':
HeeksCAD.cpp:(.text+0xf68b): undefined reference to `SetPickingColor(unsigned int)'
/usr/bin/ld: HeeksCAD.cpp:(.text+0xf75c): undefined reference to `SetPickingColor(unsigned int)'
/usr/bin/ld: CMakeFiles/heekscad.dir/MarkedList.cpp.o: in function `MarkedList::render_move_grips(bool, bool)':
MarkedList.cpp:(.text+0x938): undefined reference to `SetPickingColor(unsigned int)'
/usr/bin/ld: CMakeFiles/heekscad.dir/MarkedList.cpp.o: in function `MarkedList::ObjectsInWindow(wxRect, MarkedObject*, bool)':
MarkedList.cpp:(.text+0x1187): undefined reference to `GetPickingName(unsigned char, unsigned char, unsigned char)'
/usr/bin/ld: MarkedList.cpp:(.text+0x1264): undefined reference to `GetPickingName(unsigned char, unsigned char, unsigned char)'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/heekscad.dir/build.make:1547: bin/heekscad-1.0.0] Error 1
make[1]: *** [CMakeFiles/Makefile2:238: src/CMakeFiles/heekscad.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
==> ERROR: A failure occurred in package().
    Aborting...

Dan-Knox, 2020-03-22:
I just confirmed that #430 fixes the issue on Ubuntu 19.04. It compiles like a boss after applying the fix.

bianchifan, 2020-03-23:
Me too in confirming party ;)
THX!!

neomilium, 2020-06-14:
Fixed in master.

closed 2020-06-14 by neomilium