Heeks/heekscad #429: Build instructions empty, also no release for ubuntu 19.10

geobruce, 2019-12-26:
When I run the command : 

> sudo add-apt-repository ppa:neomilium/heekscnc-devel

I get:

> E: The repository 'http://ppa.launchpad.net/neomilium/heekscnc-devel/ubuntu eoan Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
I'm on Ubuntu 19.10 what should I do to be able to run this software in my installation?
The instructions on how to build it yourself are empty: https://github.com/Heeks/heekscad/wiki/Ubuntu:-Installation
Thanks in advance

fxrnando, 2020-06-10:
As correlated, please release an ubuntu 20.04 ppa installation package

neomilium, 2020-06-14:
@geobruce Instructions to build it under Unix-like systems can be found in README-Unix.md file. And I just refreshed wiki page about install and build under Ubuntu/Debian.

About PPA for 19.10 and 20.04, I requested a build on launchpad but I'm pretty sure that will require some patches to work due to missing 3.0 version of wxWidget on 20.04.

@danheeks : Which version of wxWidget do you use on your setup?

neomilium, 2020-06-14:
@geobruce @fxrnando I updated and patched various repo to make recent versions of whole stack, related to HeeksCNC, works on recent Ubuntu, with recent compiler. Could you give some feedback about Ubuntu 19.10 and 20.04 (which I have not tested)?