Heeks/heekscad #432: Upgrade problems 20.04

d3tic, 2020-07-10:
I upgrade to 20.04 and I am getting an error with libboost.

[image]

I have removed and reinstalled twice removed a couple of times removed other troublesome packages but I still get the same error.  After I upgraded I have a number of double packages and I stil see kubuntu flash screens even though I upgraded to standard gnome.

spiderdab, 2020-09-01:
Hi, I'm old time user of heekscad/cam, and I'm trying to compile latest version on pop os (ubuntu 20.04) and on ubuntu 18.04, but I've the same problem mentioned above:
`Python 2.7.18rc1 (default, Apr  7 2020, 12:05:55) 
[GCC 9.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import area
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: /lib/x86_64-linux-gnu/libboost_python38.so.1.71.0: undefined symbol: _Py_tracemalloc_config
>>> `
I've reproduced the error calling the import statement in python, but it happens in the software after clicking on the "GO" button to process the paths.
Can you please help on this?
The latest version in windows10 works as expected.

lpgasparotto, 2022-04-21:
This is Python related issue libboost Python component was compiled for Python 3.8 while Opencam, Libarea were compiled using Python 2.X. A fix for this is to uninstall libboost headers and to compile it for Python 2.X.

I tried to compile everything using Python 3.X for avoiding the "undefined symbol" errors but then a got a lot of syntax errors, probably due Python 2 / 3 incompatibilities.

A permanent fix would be to rewrite HeeksCad / HeeksCNC for being fully compatible with Python 3.

bullestock, 2022-08-28:
I have created a fork which has been updated for Python 3. See https://github.com/bullestock/heekscad/

lpgasparotto, 2022-08-28:
Do OpenCam and Libarea need to be compiled against Python 3 for using HeeksCNC?
Thank you very much for doing it!