Heeks/heekscad #438: OCE is discontinued, can this be made to work with OpenCascade?

frankspace, 2022-06-30:
According to [this comment](https://github.com/tpaviot/oce/issues/745#issuecomment-992285943), OCE is EOL.  I cannot get this to compile on Artix with OpenCASCADE, primarily running into repeated `fatal error: Handle_Geom_TrimmedCurve.hxx: No such file or directory` errors.  There is a "/usr/include/opencascade/Geom_TrimmedCurve.hxx" on my box from OpenCASCADE, and during compilation CMake says, among other things, "Found OCC include dir: /usr/include/opencascade".  It does complain about not finding OCE.  I simply don't know enough programming to know whether this does work with OpenCASCADE and I'm just doing something wrong, or if something needs to be done under the hood for compatibility.  Nonetheless, I don't seem to be able to get this to compile against OpenCASCADE, and it seems that with OCE being discontinued, perhaps there ought to be a way to do that?

lpgasparotto, 2022-07-31:
Hi, it seems OpenCascade uses different names for certain files. In this specific case
`/usr/include/opencascade/Handle_Geom_TrimmedCurve.hxx`
Seems to be replaced by:
`/usr/include/opencascade/Geom_TrimmedCurve.hxx`
A workaround is to create a symbolic link with the old name pointing to the current file, something like:
`ln -s /usr/include/opencascade/Geom_TrimmedCurve.hxx Handle_Geom_TrimmedCurve.hxx`
And to repeat this for every missing file. I can't guarantee it will solve all the compilation issues but at least it will hwlp to find the missing files.

bullestock, 2022-08-26:
I have created a fork which compiles with the latest OpenCascade (7.6.0). See https://github.com/bullestock/heekscad/

hadess, 2022-09-13:
> I have created a fork which compiles with the latest OpenCascade (7.6.0). See https://github.com/bullestock/heekscad/

I tried to build it against OpenCascade 7.6.2 (or rather https://gitlab.com/blobfish/occt.git/ commit `f3f00c9e9487713be0c64f36c53590cc0f30e5a6`), and ran into errors as well:
<details><summary>build log</summary>

```
[16/135] Building CXX object interface/CMakeFiles/heeksintf.dir/Geom.cpp.o
/run/build/heekscad/interface/Geom.cpp: In function ‘bool intersect(const gp_Pln&, const gp_Pln&, gp_Lin&)’:
/run/build/heekscad/interface/Geom.cpp:80:81: warning: ‘static opencascade::handle<T> opencascade::handle<T>::DownCast(const opencascade::handle<T2>&, typename std::enable_if<(! opencascade::is_base_but_not_same<T2, T>::value), void*>::type) [with T2 = Geom_Plane; T = Geom_Plane; typename std::enable_if<(! opencascade::is_base_but_not_same<T2, T>::value), void*>::type = void*]’ is deprecated: down-casting from object of the same or unrelated type is meaningless [-Wdeprecated-declarations]
   80 |                         Handle (Geom_Surface) hs = Handle (Geom_Plane)::DownCast(g_p);
      |                                                                                 ^
In file included from /app/include/opencascade/Bnd_Box.hxx:22,
                 from /run/build/heekscad/src/stdafx.h:75,
                 from /run/build/heekscad/interface/Geom.cpp:5:
/app/include/opencascade/Standard_Handle.hxx:203:19: note: declared here
  203 |     static handle DownCast (const handle<T2>& theObject, typename opencascade::std::enable_if<!is_base_but_not_same<T2, T>::value, void*>::type = 0)
      |                   ^~~~~~~~
/run/build/heekscad/interface/Geom.cpp:81:82: warning: ‘static opencascade::handle<T> opencascade::handle<T>::DownCast(const opencascade::handle<T2>&, typename std::enable_if<(! opencascade::is_base_but_not_same<T2, T>::value), void*>::type) [with T2 = Geom_Plane; T = Geom_Plane; typename std::enable_if<(! opencascade::is_base_but_not_same<T2, T>::value), void*>::type = void*]’ is deprecated: down-casting from object of the same or unrelated type is meaningless [-Wdeprecated-declarations]
   81 |                         Handle (Geom_Surface) hs2 = Handle (Geom_Plane)::DownCast(g_p2);
      |                                                                                  ^
/app/include/opencascade/Standard_Handle.hxx:203:19: note: declared here
  203 |     static handle DownCast (const handle<T2>& theObject, typename opencascade::std::enable_if<!is_base_but_not_same<T2, T>::value, void*>::type = 0)
      |                   ^~~~~~~~
[18/135] Running utility command for translations
/run/build/heekscad/translations/de/HeeksCAD.po:9: warning: header field 'Language' missing in header
/run/build/heekscad/translations/es/HeeksCAD.po:9: warning: header field 'Project-Id-Version' still has the initial default value
/run/build/heekscad/translations/es/HeeksCAD.po:9: warning: header field 'PO-Revision-Date' still has the initial default value
/run/build/heekscad/translations/es/HeeksCAD.po:9: warning: header field 'Last-Translator' still has the initial default value
/run/build/heekscad/translations/es/HeeksCAD.po:9: warning: header field 'Language-Team' still has the initial default value
/run/build/heekscad/translations/es/HeeksCAD.po:9: warning: header field 'Language' missing in header
/run/build/heekscad/translations/ko/HeeksCAD.po:2: warning: header field 'Language' missing in header
/run/build/heekscad/translations/ru/HeeksCAD.po:2: warning: header field 'Language' missing in header
/run/build/heekscad/translations/sk/HeeksCAD.po:10: warning: header field 'Language' still has the initial default value
/run/build/heekscad/translations/zh/HeeksCAD.po:7: warning: header field 'Language' missing in header
[41/135] Building CXX object src/CMakeFiles/heekscad.dir/ConversionTools.cpp.o
FAILED: src/CMakeFiles/heekscad.dir/ConversionTools.cpp.o 
/run/ccache/bin/c++ -DHAVE_CONFIG_H -DHAVE_FSTREAM -DHAVE_IOMANIP -DHAVE_IOSTREAM -DHAVE_LIMITS_H -DHEEKSCAD -DLIN -DLININTEL -DTIXML_USE_STL -DUNICODE -DUSE_GENERIC_TREECTRL -DWXUSINGDLL -D_OCC64 -D__WXGTK3__ -D__WXGTK__ -I/run/build/heekscad -I/run/build/heekscad/src -I/usr/include/python3.10 -I/app/include/area -isystem /app/lib/wx/include/gtk3-unicode-3.2 -isystem /app/include/wx-3.2 -isystem /app/include/opencascade -O2 -g -pipe -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -pthread -pthread   -Wall -m64 -Wall -m64 -MD -MT src/CMakeFiles/heekscad.dir/ConversionTools.cpp.o -MF src/CMakeFiles/heekscad.dir/ConversionTools.cpp.o.d -o src/CMakeFiles/heekscad.dir/ConversionTools.cpp.o -c /run/build/heekscad/src/ConversionTools.cpp
/run/build/heekscad/src/ConversionTools.cpp: In function ‘bool ConvertSketchToEdges(HeeksObj*, std::__cxx11::list<std::vector<TopoDS_Edge> >&)’:
/run/build/heekscad/src/ConversionTools.cpp:407:64: error: ‘Caught’ is not a member of ‘Standard_Failure’
  407 |                 Handle(Standard_Failure) e = Standard_Failure::Caught();
      |                                                                ^~~~~~
/run/build/heekscad/src/ConversionTools.cpp: In function ‘bool ConvertEdgesToFaceOrWire(const std::vector<TopoDS_Edge>&, std::__cxx11::list<TopoDS_Shape>&, bool)’:
[... cut ...]
/run/build/heekscad/src/DigitizeMode.cpp: In member function ‘DigitizedPoint DigitizeMode::digitize1(const wxPoint&)’:
/run/build/heekscad/src/DigitizeMode.cpp:281:16: warning: variable ‘best_dp’ set but not used [-Wunused-but-set-variable]
  281 |         double best_dp = 0;
      |                ^~~~~~~
[56/135] Building CXX object src/CMakeFiles/heekscad.dir/CxfFont.cpp.o
```

</details>

That was after I modified heekscad to use `BRepAlgoAPI_Fuse` instead of `BRepAlgo_Fuse` which got removed from the 7.6.0 API.